micro-ROS/system_modes #8: Minimize redundancy between roslaunch and system-modes model file

ralph-lange, 2020-01-24:
For launching the nodes specified in a system-mode models file (SHM file) for control by the mode-manager, all these nodes have to be specified additionally in a launch file. This redundancy is error-prone and should be avoided or at least minimized.

One idea is to provide a convenience function for launch that launches all nodes specified in an SHM file.

norro, 2020-02-13:
A probably quite convenient ways would be to provide a LaunchDescription that reads an SMH file and starts up all nodes and a mode-manager accordingly, see #11.
Does this sound reasonable to you, @ralph-lange?

ralph-lange, 2020-02-14:
This would require that the SMH file includes the necessary details on the packages, which nodes to deploy per OS process, ... Basically, this would make the SMH file format another launch file format. I would avoid this - despite my last sentence in the first message of this thread.