micro-ROS/system_modes #9: How to add the system_modes manager into a launch file?

jginesclavero, 2020-01-29:
Hi! 
I'm trying to add the mode-manager node into a launch and I couldn't get it. I think that is related to how the node reads the modes file from arguments.
Do you have any example of a launcher file that launch the mode-manager?

Thank you so much!

ralph-lange, 2020-01-29:
Hi @jginesclavero, the normal launch_ros.actions.Node description does not work since the path to the model file is currently passed as plain command line argument and not as ROS parameter. This should be fixed.
Until then, you may launch the node by a launch.actions.ExecuteProcess description. I created a small example at https://github.com/micro-ROS/system_modes/blob/feature/launch_file_example/system_modes_examples/launch/example_mode_manager.launch.py 

Check out the branch feature/launch_file_example, build it, and then run it `ros2 launch system_modes_examples example_mode_manager.launch.py`

ralph-lange, 2020-01-29:
@norro, it should be possible to pass the path to the SHM file as ROS parameter - either instead of the normal command line argument method or as alternative method.

jginesclavero, 2020-01-30:
Ok! Thank you so much for your answer @ralph-lange. Including it in my launchers :+1:

ralph-lange, 2020-01-30:
Hi @jginesclavero, I added a ROS parameter modelfile to the mode_manager node and the mode_monitor node. (Note the underscore in the node names.) This allows passing the path to the SHM model file using ROS parameters. Furthermore, I created generic launch files for both nodes in the system_modes package and modified the example in system_modes_example/launch to use them.

marinatto, 2020-01-30:
Hey guys, I was trying to add this same node into a ROS2 launch file yesterday, and I could make it using the launch_ros.actions.Node. So far, I didn't notice anything wrong until now. The only problem I had to handle was  `Error parsing command line: too many positional options have been specified on the command line`, I couldn't set a node_name neither a node_namespace. I don't know if it's intentional though. For the arguments, I used os.path.join from os library + `from ament_index_python.packages import get_package_share_directory` to get the path from the yaml file and passed it as the argument.

ralph-lange, 2020-01-30:
Hi @marinatto, could you please repeat your test with my latest changes in the branch feature/launch_file_example?

marinatto, 2020-01-30:
Hi @ralph-lange, it worked just fine but I still get the same error when I try to set  `node_namespace` or `node_name`

ralph-lange, 2020-01-31:
Hi @marinatto, I was able to use node_name with Eloquent. The difference to Dashing is that the renaming arguments are passed behind the new --ros-args option, which I now treat explicitly in the boost::options instance in mode_manager_node.cpp (cf. https://github.com/micro-ROS/system_modes/blob/feature/launch_file_example/system_modes/src/mode_manager_node.cpp#L67).
To fix the issue under Dashing, I will remove the use of boost::options completely in the next days and rely solely on ROS parameters.

norro, 2020-02-11:
@jginesclavero Can you have a look at #11 and the exemplary file [feature/launch_file_example:launch/example_system.launch.py](https://github.com/micro-ROS/system_modes/blob/feature/launch_file_example/system_modes_examples/launch/example_system.launch.py), if this meets your expectations?
This works for eloquent, I will have a look at dashing soon.

norro, 2020-02-13:
@jginesclavero I am closing this issue as an example is now included. Feel free to re-open or create a new one if it does not match your expectations.

closed 2020-02-13 by norro