micro-ROS/system_modes #10: Make ModeManagerNode and ModeMonitorNode themselves lifecycle nodes?

ralph-lange, 2020-02-07:
- Seems natural to use the same concept (i.e. runtime lifecycle) for these to nodes themselves.
- Yet, in many applications, the mode manager node will be used continuously from system start to shutdown so that there is no need for a runtime lifecycle
- On the other hand, one might think of a distributed system with multiple mode managers for subsystems which could be deactivated during runtime.
- If making the two nodes to lifecycle nodes, convenience launch files with auto-activate functionality should be provided.

norro, 2020-02-11:
Will definitely have a look at the suggested auto-activate launch functionality, which seems to be a preconditions. Didn't yet manage to implement this, see #11.

norro, 2021-07-23:
Since starting mode manager and mode monitor via launch files is the recommended way anyway (see [system modes examples](https://github.com/micro-ROS/system_modes/tree/master/system_modes_examples)), this should be easily doable and comvenient by now.