micro-ROS/system_modes #24: Redundant service call parameter "node_name" for system_modes services

chcorbato, 2020-07-17:
Currently, `system_modes/srv/GetAvailableModes`, `system_modes/srv/ChangeMode` and `system_modes/srv/GetMode` require as parameter the `node_name`. However, the similar services from `lifecycle_msgs/srv/*` do not require to pass that parameter, since the service name/scope already includes it.
 For example:
```
ros2 service call /actuation/get_available_modes system_modes/srv/GetAvailableModes "{node_name: actuation}"}"
```

Maybe that service parameter could be optional, also for system_modes to have a more similar interface to lifecycle?

norro, 2020-07-21:
Thanks, @chcorbato, this is a very valuable observation. This redundancy was indeed also present in the according `lifecycle_msgs/srv/` when we first started this project (before pushing it to github), but was since removed: https://github.com/ros2/rcl_interfaces/pull/39 (October 2018).

So you are right, this should be removed from the system mode services as well.

norro, 2020-07-23:
@chcorbato please have a look at https://github.com/micro-ROS/system_modes/issues/24

closed 2020-09-15 by norro

marioney, 2020-10-15:
Hi @norro 

This issue is still present in the [/feature/rules](https://github.com/micro-ROS/system_modes/tree/feature/rules) branch, I'm not sure if a new issue is needed, but It'll be good to apply the solution to that branch as well.

norro, 2020-10-15:
Very good point, I will port it there as well.

norro, 2020-10-16:
@marioney should be ported now. If it causes any issues, please don't hesitate to open further bug reports.