micro-ROS/system_modes #26: system_modes_examples problem to change node status

chcorbato, 2020-07-17:
In the `system_modes_example`, after activating the `actuation` system, I am trying to change the state of `driver_base`

Also, I noticed that when I check all available states, I get all the states specified by the [lifecycle model in ROS2](http://design.ros2.org/articles/node_lifecycle.html). However, I am not sure all the states and transitions are implemented for [`drive_base`](https://github.com/micro-ROS/system_modes/blob/master/system_modes_examples/src/drive_base.cpp) for example.
Is this something that needs improving? Could this be interesting in the context of MROS?

FranFin, 2020-07-20:
@chcorbato Hi! Thank you for your interest in micro-ROS. Notice that we’ll discuss the recent contributions and developments in the next [Embedded WG meeting](https://discourse.ros.org/t/ros-2-embedded-wg-meetings/15460).
You’re invited to bring your micro-ROS related issue to this platform. We’ll be very happy to have you join us and discuss your questions with the community.

chcorbato, 2020-07-20:
Thanks @FranFin . I am not into the "embedded" world, but am in collaboration with @norro and @ralph-lange to apply system modes in another project called [MROS](https://robmosys.eu/mros/)

Regarding this issue, I was using incorrect transitions IDs, apologies. I close the issue.

closed 2020-07-20 by chcorbato