micro-ROS/system_modes #44: New release into ROS 2 Rolling?

clalancette, 2020-09-21:
ROS 2 Rolling has completely removed the `set_on_parameters_set_callback` method, so the package is currently failing to build on Rolling: http://build.ros2.org/view/Rbin_uF64/job/Rbin_uF64__system_modes_examples__ubuntu_focal_amd64__binary/107/console .

The good news is that it looks like this is fixed on master.  Would you mind doing a new release into Rolling to fix the build there?  Thanks!

norro, 2020-09-22:
Hi @clalancette, I will happily give it another go. The thing is, the release infrastructure changed for rolling, see https://github.com/ros2-gbp/system_modes-release/issues/1 and is not really ready, yet.
But I am eager to try it as well, so I will get in contact with Steven again.

norro, 2020-09-30:
Just triggered a new rolling release of version 0.3.0, let's see how it works out.

closed 2020-09-30 by norro