micro-ROS/system_modes #45: Inference issue

jginesclavero, 2020-09-23:
Hi @norro ! I'm testing the mode inference into [Pilot-URJC](https://github.com/MROS-RobMoSys-ITP/Pilot-URJC) and I dont know how it works. [This is my modes file](https://github.com/MROS-RobMoSys-ITP/Pilot-URJC/blob/master/pilot_urjc_bringup/params/pilot_modes.yaml). I bring up the system and set the NORMAL mode `ros2 service call /pilot/change_mode system_modes/ChangeMode "{mode_name: 'NORMAL'}"`.

[image]

At this point, I set the `laser_resender` node into a `inactive` mode. This node must be `active` node diuring the `NORMAL` mode. I would hope that the inference mode would take this information and inferred that the correct mode in this situation is the `DEGRADED` mode, the only one in which the `laser_resender` is in the `inactive` mode. Instead of the above, nothing happens.

[image]

Am I understanding correctly how the inference works?

norro, 2020-09-28:
I will have a look at this today.

norro, 2020-09-28:
I agree with your expectation. However, I am wondering why the monitor does not show *any* inference for the `pilot` system at all: `state actual` and `mode actual` is empty in both of your screenshots, which shouldn't be the case.

A reason might be that it doesn't seem to know anything about the `pointcloud_to_laser`.
@jginesclavero Was the `pointcloud_to_laser` node running when you did the screenshots and was the mode monitor running right from the start so that it could observe all lifecycle and parameter events?

jginesclavero, 2020-09-29:
> A reason might be that it doesn't seem to be able to infer the mode for the laser_resender, maybe it missed the parameter event of laser_resender_test param.

Yes, this param does not exist. What is the syntax of a sub-system in the modes file if you don't want to manage any parameter of this sub-system?

> Was the laser_resender node running when you did the screenshots and was the mode monitor running right from the start so that it could observe all lifecycle and parameter events?

I think so, let me check it again.

To clarify, I have used `master` branch to do this tests. I will test `feature/rules` branch in this morning.
Thank you for your help.

norro, 2020-09-29:
I see!
Currently, there has to be a parameter, because otherwise the mode is invisible from the outside and can't be inferred in any way.
 
You may want to introduce a simple *fake* parameter, that is a bool for example or - to make it a bit easier to understand - a string parameter that holds the name of the mode. We stumbled upon this question before, /cc @ralph-lange.

norro, 2020-10-02:
Is this issue still valid or can it be closed?

jginesclavero, 2020-10-02:
We can close it.

closed 2020-10-02 by jginesclavero