micro-ROS/system_modes #48: Layered handling of node and (sub-)system errors

chcorbato, 2020-09-30:
from (#47 )
> 

_continuation_

norro, 2020-09-30:
What the mode manager will actually already sense is the deviation between the requested state/mode and the actual state/mode. This is not yet merged to master, but available in the `feature/rules` branch, because it is necessary in order to decide when to apply rules. See [feature/rules:mode_inference.cpp](https://github.com/micro-ROS/system_modes/blob/feature/rules/system_modes/src/system_modes/mode_inference.cpp#L657-L674).
Reporting these deviations to diagnostics is an interesting idea.

This is again a question of timing, though. When a state/mode transition is requested, there is *always* and *immediately* a deviation, since systems/nodes will take *some* time to perform the transition. So the mode manager will have to decide, when to report the deviation, i.e. when to assume that the transition takes to long and the deviation therefore can be considered an erroneous deviation. Do you have an idea how/when to do this? After half a second? A second? ... @chcorbato

norro, 2020-09-30:
Suggestion:

1. When a deviation is detected, wait a certain time `t_0` before considering it an erroneous deviation
1. After `t_0`, try to apply a rule, if an appropriate rules exists. If no rule exists, try to recover the node/system
1. Wait a certain time `t_1` and if nothing happened, report the erroneous deviation, e,g., through diagnostics
/cc @chcorbato @ralph-lange

chcorbato, 2020-09-30:
I like very much your suggestion of a configurable time limit for each management layer!

norro, 2020-09-30:
@chcorbato The `feature/rules` is merely a micro-ROS  experiment by now btw. For "2. If it does not succeed [...] tries to recover from the error using rules" I consider metacontrol (reconfiguration actions?) in charge.
We are even happy to drop the system modes rules feature completely once the metacontrol part for this task is integrated with system modes.

chcorbato, 2020-09-30:
I see. Currently @jginesclavero is trying to get results with that feature this week, by adding such a rule in Pilot-URJC system model.

I propose we keep this test for this week and analyse the result afterwards (usefulness, problems...) to then make an informed decision to move the feature to the metacontrol part.

What do you think @norro @jginesclavero ?
@norro are you available to keep supporting @jginesclavero on this today and tomorrow?

norro, 2020-09-30:
Yes, I am available today and tomorrow to help with upcoming issues.

jginesclavero, 2020-09-30:
Hi @norro @chcorbato !
I was testing the `feature/rules` branch and it works as we expected. In short, I have defined a rule that changes to `DEGRADED` mode (navigation with pointcloud_to_laser) if the laser_driver is not in `active` state. The mode is changed immediately, works really nice. 
I have done some navigation tests where I force a laser failure and the mode change correctly, the laser is replaced by the pointcloud and the navigation continues.

marioney, 2020-09-30:
> Do you have suggestions for these times in the case of navigation2 @fmrico @marioney @jginesclavero @lbajo ?

From the metacontroller point of view, the reasoning cycle is very slow (about 2 sec) so we're safe with half of that I guess.  I'm not sure how that time affects the navigation 2, but I'm guessing it does not.

norro, 2021-07-23:
Closing this issue soon as it has successfully been shown in the MROS pilots.