micro-ROS/system_modes #60: Prevent redundant mode changes

norro, 2021-03-01:
Ensure that redundant mode changes are handled properly.
If mode change is requested to a mode that is already active according to the mode inference, no service should be called, but the ModeChange service request should still be returned with success.

YoYoChaico, 2021-03-01:
Hey! 
Where can i find the code for it?

norro, 2021-03-02:
I expect this to be handled in the mode manager in [change_mode()](https://github.com/micro-ROS/system_modes/blob/master/system_modes/src/system_modes/mode_manager.cpp#L416) and [change_part_mode()](https://github.com/micro-ROS/system_modes/blob/master/system_modes/src/system_modes/mode_manager.cpp#L519).
Before doing anything in these methods, mode manager should compare the requested mode with the current mode, accessible from the mode inference.

closed 2021-04-06 by norro