micro-ROS/system_modes #69: Approach to make named mode default

nielsvd, 2021-04-07:
Assuming my current understanding of this package is sufficient. Let us say (for simplicity) we want to assign two modes to an `example_node`: mode `A` and mode `B`. In both modes, we would like to redefine a long list of parameters. The package needs a `__DEFAULT__` mode to be defined that is actually named as such, assume that we choose mode `A` to be default, the current working approaches that I found is to either:
1. Rename mode `A` to `__DEFAULT__`, meaning our `example_node` has modes `__DEFAULT__` and `B`. Effective, but not very idiomatic.
2. Define an additional `__DEFAULT__` mode, meaning our `example_node` has modes `A`, `B`, `__DEFAULT__`. To ensure that `A` and `__DEFAULT__` are effectively the same, it is necessary to keep the parameters in their configuration are identical. Especially when there are many parameters this becomes tedious.

I'm not sure how to do this most effectively. But it would be nice if we can define modes `A` and `B` and choose the default in some other way, perhaps through another key in the yaml file (next to `type` and `modes`)?

norro, 2021-04-14:
@nielsvd Would you prefer that
1. the first mode defined for a part (system or node) in the SMH file is considered the default mode or 
2. at least one of the modes defined for a part (system or node) in the SMH file has to be marked default?

nielsvd, 2021-04-15:
I actually think both options would be fine. Perhaps 2, with a fall-back to 1 in case the explicit choice is missing (also for backwards compatibility)?

closed 2021-07-21 by norro