micro-ROS/system_modes #95: [FEA] Only config subset of nodes in a certain mode.

ZhenshengLee, 2022-10-26:
Hi, first of all I want to say this is a great project to manage lifecycle nodes!

There'a a need to config the lifecycle state of only a subset of nodes in one mode, but config another subset of nodes in other modes.

Let's say, can the following config work? (Currently I don't need the extra parameter-based modes either)

```yaml
# system modes example
---

actuation:
  ros__parameters:
    type: system
    parts:
      manipulator
      drive_base
    modes:
      __DEFAULT__:
        manipulator: inactive
        drive_base: inactive
      MODERATE:
        manipulator: active
      PERFORMANCE:
        drive_base: active
```

Thanks!

ralph-lange, 2022-11-03:
I assume this issue is superseded by https://github.com/micro-ROS/system_modes/issues/96, or?

ZhenshengLee, 2022-11-04:
@ralph-lange thank you again!

> I assume this issue is superseded by https://github.com/micro-ROS/system_modes/issues/96, or?

No, this case is different with #96 

In #96 , the bug/fea is about changing mode state to set one of nodes to inactive.

In this issue, the bug/fea is about chaning mode state to set only one node to other state, and leave other nodes not changing their state(remain the same)