micro-ROS/system_modes #96: [FEA] Allow deactivating nodes when system is activated.

ZhenshengLee, 2022-10-27:
There is a need to selectively disable the function of a node, that's say deactivating a node in a certain system mode.

Should the following config work? Thanks!

```yaml
---

actuation:
  ros__parameters:
    type: system
    parts:
      drive_base
      left/manipulator
      right/manipulator
    modes:
      __DEFAULT__:
        drive_base: active.__DEFAULT__
        left/manipulator: inactive
        right/manipulator: inactive
      MODERATE:
        drive_base: active.__DEFAULT__
        left/manipulator: inactive
        right/manipulator: active
      PERFORMANCE:
        drive_base: active.__DEFAULT__
        left/manipulator: active.__DEFAULT__
        right/manipulsator: inactive

left/manipulator:
  ros__parameters:
    type: node
    modes:
      __DEFAULT__:
        ros__parameters:
          max_torque: 0.1
      WEAK:
        ros__parameters:
          max_torque: 0.1
      STRONG:
        ros__parameters:
          max_torque: 0.2

drive_base:
  ros__parameters:
    type: node
    modes:
      __DEFAULT__:
        ros__parameters:
          max_speed: 0.1
          controller: PID
      SLOW:
        ros__parameters:
          max_speed: 0.2
          controller: PID
      FAST:
        ros__parameters:
          max_speed: 0.9
          controller: MPC

right/manipulator:
  ros__parameters:
    type: node
    modes:
      __DEFAULT__:
        ros__parameters:
          max_torque: 0.11
      WEAK:
        ros__parameters:
          max_torque: 0.05
      STRONG:
        ros__parameters:
          max_torque: 0.3
```

ralph-lange, 2022-11-03:
At first glance, this should work. You make even simply write `drive_base: active` instead of `drive_base: active.__DEFAULT__` (as in `right/manipulator: active`). Did you experience any problems with this configuration?

ZhenshengLee, 2022-11-04:
@ralph-lange  thank you for your fast reply!

> At first glance, this should work

In currently design, In any specific MODE(MODERATE and PERFORMANCE), all node must be in ACTIVE state, if any of the nodes become INACTIVE, the mode_inference will think the MODE is in DEACTIVATING state

> Did you experience any problems with this configuration?

- The segfault happens with mode_manager with this config-yaml

I will repeat the test and give more details.