micro-ROS/system_modes #97: [FEA] Allow all inactive in DEFAULT MODE.

ZhenshengLee, 2022-11-12:
#### SHM file

```yaml
# system modes example
---

actuation:
  ros__parameters:
    type: system
    parts:
      drive_base
      left/manipulator
      right/manipulator
    modes:
      __DEFAULT__:
        drive_base: inactive.__DEFAULT__
        left/manipulator: inactive.__DEFAULT__
        right/manipulator: inactive.__DEFAULT__
      LEFT:
        # drive_base: active.__DEFAULT__
        left/manipulator: active.__DEFAULT__
        # right/manipulsator: inactive.__DEFAULT__
      RIGHT:
        # drive_base: active.__DEFAULT__
        # left/manipulator: inactive.__DEFAULT__
        right/manipulator: active.__DEFAULT__

left/manipulator:
  ros__parameters:
    type: node
    modes:
      __DEFAULT__:
        ros__parameters:
          max_torque: 0.1

drive_base:
  ros__parameters:
    type: node
    modes:
      __DEFAULT__:
        ros__parameters:
          max_speed: 0.1
          controller: PID

right/manipulator:
  ros__parameters:
    type: node
    modes:
      __DEFAULT__:
        ros__parameters:
          max_torque: 0.11

```

#### step to reproduce

```sh
zs@zs-vm-2204:~/zs_ws/ga_ros$ time ros2 service call /actuation/change_state lifecycle_msgs/ChangeState "{transition: {id: 1, label: configure}}"
waiting for service to become available...
requester: making request: lifecycle_msgs.srv.ChangeState_Request(transition=lifecycle_msgs.msg.Transition(id=1, label='configure'))

response:
lifecycle_msgs.srv.ChangeState_Response(success=True)

real    0m2.185s
user    0m0.459s
sys     0m0.086s
```

#### expected behavior

all nodes would be in DEFAULT mode, which is all inactivated

#### actual behavior

[image]

ralph-lange, 2022-11-16:
@ZhenshengLee, sorry for answering so late/slow on your issues. I'll analyze them in next week. Looks as if they are related, i.e., have the same root-cause.

ZhenshengLee, 2022-11-28:
> @ZhenshengLee, sorry for answering so late/slow on your issues. I'll analyze them in next week. Looks as if they are related, i.e., have the same root cause.

Friendly ping...